Ultimaker/Cura #16253: modern exotic sports car

cuppm, 2023-07-24:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Anycubic Kobra 2

### Describe your model

sports car from thingiverse: https://www.thingiverse.com/thing:4966156

### Add your .zip here ⬇️

[CFFFP_modern_exotic_sports_car_fixed.zip](https://github.com/Ultimaker/Cura/files/12140480/CFFFP_modern_exotic_sports_car_fixed.zip)

GregValiant, 2023-07-24:
Thanks for the report and the file.
I cannot get Cura 5.4.0 to slice this model.  It sliced in 4.13.1 and 5.2.2.
I'll mark this as a duplicate report and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic