Ultimaker/Cura #16255: Add iNSTONE profiles

ITProggeruz, 2023-07-24:
### Is your feature request related to a problem?

`Hi` everyone i want iNSTONE profiles

—

<details><summary>Because</summary>
<p>

I got a `NEW` 3D printer
And it’s iNSTONE Inventor pro
The support site is down which it has drivers for connecting
You can give me tips to profile settings until the presets will adds

</p>
</details> 

Thanks, please don’t make me sad by not responding

### Describe the solution you'd like

I can solve it by asking from you guys and set up
But it it is a suggestion and issue

### Describe alternatives you've considered

Any people who can help me

### Affected users and/or printers

Maybe all of instone printers

### Additional information & file uploads

I don’t have any resources to download drivers, control app, e.t.c..

`Or you can give me link to download and it’s OPTIONAL`

ITProggeruz, 2023-07-24:
The operation system and other stuff is:
/

ITProggeruz, 2023-07-24:
And it somehow closed this issue

ITProggeruz, 2023-07-24:
Hoping this issue will complete in a short time

GregValiant, 2023-07-24:
Are they still in production?  All I found was a small printer that is no longer available (and I've never heard of them prior to this).

ITProggeruz, 2023-07-24:
no, its out of stock (i brought it not from online)

ITProggeruz, 2023-07-24:
also the websit is closed (idk owned or hacked)

ITProggeruz, 2023-07-24:
socials are facebook and twitter
and if you look up from wayback machine the download files don't exist anymore

ITProggeruz, 2023-07-24:
wait also i changed places of some details

ITProggeruz, 2023-07-24:
> Are they still in production? All I found was a small printer that is no longer available (and I've never heard of them prior to this).

its not on popular marketing sites

ITProggeruz, 2023-07-24:
oh wait also someone used this product before me

GregValiant, 2023-07-24:
The UltiMaker Company will not be putting money into developing Cura support for an obsolete printer built by another manufacturer.  This is something you will have to do for yourself.
You can try installing a similar printer (look at the Weefun Tina2 printer).  Then you can go to the Machine Settings and adjust for your printer.  You could also install it as a Custom FFF printer.
I'll go ahead and close this.  I hope you understand.

closed 2023-07-24 by GregValiant

ITProggeruz, 2023-07-24:
how i can create FDM printers then???

ITProggeruz, 2023-07-24:
let me check in issues about fdm printer

ITProggeruz, 2023-07-24:
it gave me 56% chance of help