Ultimaker/Cura #16258: Cutting / Infill Mesh Cylinders Not Saving Position Between Loads

BigWampa1989, 2023-07-24:
### Cura Version

5.4.0 and 5.2.1

### Operating System

Windows 10 - Klipper

### Printer

Creality Ender 3 

### Reproduction steps

When I insert a cylinder (and other objects as well) and use them as cutting or infill meshes when I save the project and then reload the project, they drop down to the build plate. Interestingly, using a square shape it retains its Z coordinate but the cylinders drop down. This includes other regular models as well. I took the print model in this file and raised it above the build plate and when I either opened it as a project or imported the models it dropped the model back down to the build plate instead of keeping its Z Coordinate. The attached screen shots show what the model is supposed to look like versus what it looks like when I open the file up again. 

### Actual results

See Above. 

### Expected results

See Above. 

### Add your .zip and screenshots here ⬇️

[Dial Gauge Holder.zip](https://github.com/Ultimaker/Cura/files/12146728/Dial.Gauge.Holder.zip)
[image]
[image]

Asterchades, 2023-07-25:
There's a General setting called rather informatively "Automatically drop models to the build plate" which is causing this. With it enabled any _individual_ model which is loaded will, as expected, be placed on the build plate - and that's exactly what's happening here.

You have two choices:
- Disable the aforementioned setting. Everything loaded will stay put at the Z height it was saved as until you move them yourself.
- Prior to saving, select all of the models when they're appropriately located and Group them (right-click option, or Ctrl+G). This will cause their positions to be stored relative to each other as a whole, rather than as individual objects. NB You will need to ungroup them if you wish to move a single piece later (don't forget to re-group after).

GregValiant, 2023-07-25:
@BigWampa1989 are you good on this now?  If it is resolved then I will go ahead and close this.

BigWampa1989, 2023-07-25:
Yes I marked it complete

closed 2023-07-25 by BigWampa1989