Ultimaker/Cura #16260: View/Edit profile only works once

FlashMoonUK, 2023-07-24:
### Cura Version

5.4.0

### Operating System

Mac OS 13.4.1

### Printer

Dremel 3d45 with plugin

### Reproduction steps

Cmd J to see profile 
close profile
Cmd J to see profile - will not longer show profile

### Actual results

the second time you wish to look at a profile the view/edit window will not open

### Expected results

should be able to use view edit multiple time

### Add your .zip and screenshots here ⬇️

nothing to add

MariMakes, 2023-08-08:
Hey @FlashMoonUK,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

I tried to follow your steps on my Windows 11 machine, but I cannot reproduce it. 
[image]

Did your profile have a specific name?
Did it have a specific setting change?
Was it a new profile or a duplicate from an already existing one? 

I'm not sure how to help if we cannot reproduce it, so any information would be valuable 🙏

FlashMoonUK, 2023-08-08:
Hi Mariska

Thanks for looking into this for me 

I have only been able to recreate this once - (today) - but that is it and I can’t get it to do it again.

I’m using a Mac with an M1 chip set

If I can get it to reproduce I will record it and send you the details

Kind regards

Chris

MariMakes, 2023-08-10:
Thanks for the update 👍 
Please let us know if you know more. 😉

closed 2023-08-24 by github-actions[bot]