Ultimaker/Cura #16261: Suggesting to adding custom FDM printers

ITProggeruz, 2023-07-24:
### Is your feature request related to a problem?

Yes and No: For my FDM iNSTONE Inventor PRO (the terms are ended); but my printer is still working well

### Describe the solution you'd like

Adding Custom FDM Printer Profile

### Describe alternatives you've considered

It is request. _**Pain inside because i do not have idea what to do with this printer (make)**_

### Affected users and/or printers

Devs, `im not nerd so who can, who does`

### Additional information & file uploads

Also i am from a town (i mean sometimes the electricity will be down **because it is so sunny** )

ITProggeruz, 2023-07-24:
this request is not important, but useful

MariMakes, 2023-09-11:
Hey @ITProggeruz,

Sorry, it took us a while to get back to you 😞 

Since Cura is open source, you or anyone else is more than welcome to work on adding this 3d printer and create a pull request yourself. You can find instructions on how to add a new 3D printer definition here: https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura 

If you are stuck in any way, please reach out! I'll be happy to help out.

ITProggeruz, 2023-11-13:
Ive tried but tho guys couldnt find a printer