Ultimaker/Cura #16263: Setting to change the default model position when loading a stl file

EmJay276, 2023-07-24:
### Is your feature request related to a problem?

No

### Describe the solution you'd like

A setting to set the default model position when loading a stl file

### Describe alternatives you've considered

Moving the model everytime by hand.

### Affected users and/or printers

All

### Additional information & file uploads

I have a fairly large bed (300x600mm) which is devided in 2 beds (front, back, each 300x300mm). By default I want to print on the center of the front bed and not the seam between the beds.

GregValiant, 2023-07-25:
Have you considered installing another printer in Cura with the bed at 300 x 300?  You could use that "Small" printer definition for most things and when you get a model that needs the big bed use the "Big" printer definition.  Other than the Machine Settings "Y(depth)" everything else would be the same.  It's something you could do now instead of waiting for a feature request to go through.
In regards to the feature request, it may not even be possible because some model files have "location" information within the file.  That would need to be dealt with.

EmJay276, 2023-07-25:
@GregValiant thanks for your feedback. 
Using a 300x300 printer seems to be a current workaround for me. 
But maintaining 2 printers with the same config will go wrong at one point.

To my knowledge cura centers the object by default to its center of gravity, at least when I use the CuraEngine via console.
Therefore adding configuration values for X, Y and Z(?) to move it to, instead of the center of the bed seems to be doable without breaking anything.

If you now where to the stl import is done, I can try to create a PR. But I hav'nt found it yet.

GregValiant, 2023-07-25:
I know just enough about the inner workings to be dangerous.
The model comes in at the XY center-of-geometry of the "bounding box" that contains the part, and at the lowest point in the Z.  The Cura Bounding Box is always square to the world (if the part is rotated, the bounding box changes size).

Files that have location data in them (not many STL's do) are respected and if there is location data then Cura will put the part at the location defined in the file.  (I believe that is used when parts are "merged".)

I maintain three printers although I only own one.  I wanted different startup gcode for PETG and for TPU so it seemed a simple workaround.

5axes, 2023-07-25:
Hello,
As this is a very specific request, there is little chance that it will ever succeed. One solution would be to use @fieldOfView  MeshTool plug-in. This plugin already has an option for randomly positioning parts on the bed to avoid always printing on the same area... All you'd have to do is replace the random position with a fixed position to be defined in the settings. 

https://github.com/fieldOfView/Cura-MeshTools/issues

[image]

EmJay276, 2023-07-25:
Thanks, I will have a look at this plugin.

5axes, 2023-07-25:
You can try by changing : 
[image]
 into 
[image]

Of course a real modification with new options in the plugin will be nicer and of course a PR as it could be usefull for some people.