Ultimaker/Cura #16264: Preview in Cura 5.4.0 is not the same as in reality.

ISMArtinz, 2023-07-24:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Creality Ender-3 V2 with bltouch

### Reproduction steps

Translate to English:
1. Design a cover for the switches in Fusion 360.
2. Pass it to the laminator.
3. Rotate the object 180º so that the thinner edges touch the print plate.
4. Print.

### Actual results

 After printing the first layer, which consists of two lines of the skirt and four lines (2 outer walls and 2 inner walls), there is a noticeable gap in the corners that shouldn't be there.

### Expected results

What should have happened is that the corners should not have any gap.

### Add your .zip and screenshots here ⬇️

[Cura.zip](https://github.com/Ultimaker/Cura/files/12149206/Cura.zip)

GregValiant, 2023-07-25:
Thanks for the report and the files.
This is the preview in Cura of the first layer.
[image]

This is the gcode from that slice read into AutoCad to take Cura out of the loop.  I know it's hard to see but if you zoom in you will see that all the lines are right where they should be.  This shows layer 1 (with the skirt turned off) up through layer 50.  They are right on top of one another.
[image]

When the nozzle moves around a curve the plastic must adhere to the bed or it will pull to the inside of the curve.  The plastic wants to stay straight and if the bed adhesion isn't sufficient then you can get what you see here.
[image]

Here is the first layer on my Ender 3 Pro.
[image]

So I don't see a bug here but rather regular old bed adhesion problems.  If you have your E-steps calibrated then you can try a couple of things.  Setting the "Initial Layer Flow" to 105% (or even 110%) is a quick and dirty fix.  If you have ABL you might want to tweak your Zoffset.

I'm going to take the bug label off this but I'll leave it open for a bit so you can respond.  I hope you understand.

ISMArtinz, 2023-07-25:
Thank you very much for the help. I thought it was an error, but it turned out to be the straps that had loosened without me realizing.

GregValiant, 2023-07-25:
The machines work hard.  Regular maintenance is a good idea so these things don't slip by.

closed 2023-07-25 by GregValiant