Ultimaker/Cura #16266: Não fatia ao girar 90º

kalvincaetano, 2023-07-24:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Ender 3 v1

### Name abnormal settings

ao girar ele em 90º, ele não fatia e preciso imprimir com giro questão de esforço fisico 

### Describe model location

centralizado com giro 90º

### Describe your model

se deixar sem girar ele fatia

### Add your .zip here ⬇️

[CE3E3V2_xbox_controller_stand2.zip](https://github.com/Ultimaker/Cura/files/12154763/CE3E3V2_xbox_controller_stand2.zip)

GregValiant, 2023-07-25:
Thanks for the report.
When I changed the Initial Layer Height to 0.25 it sliced.
I'll mark this as a duplicate report as there have been a lot of reports of this behavior.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic