Ultimaker/Cura #16271: super lent a calculer la découpe ?

fabnuss, 2023-07-25:
### Cura Version

5.4.0

### Operating System

win 10

### Printer

alfawise u20

### Reproduction steps

je découpe une piece par cura et il met un temps pas possible, casiment 20 min pour calculer un socle, c'est n'importe quoi, j'ai une ancienne version qui 1 à 2 min ???
pourriez vous aussi faire en sorte de traduire cette page-ci en francais car on comprends rien, moi je parle pas anglais, merci

### Actual results

je découpe une piece par cura et il met un temps pas possible, casiment 20 min pour calculer un socle, c'est n'importe quoi, j'ai une ancienne version qui 1 à 2 min ???
pourriez vous aussi faire en sorte de traduire cette page-ci en francais car on comprends rien, moi je parle pas anglais, merci

### Expected results

je découpe une piece par cura et il met un temps pas possible, casiment 20 min pour calculer un socle, c'est n'importe quoi, j'ai une ancienne version qui 1 à 2 min ???
pourriez vous aussi faire en sorte de traduire cette page-ci en francais car on comprends rien, moi je parle pas anglais, merci

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-07-28:
I have noticed that when there are a lot of features packed onto the same plane (heavily grained wood floors, tile floors, the top surface of your model, etc) that slicing can take a really, really long time.  I think the Cura team is aware of the problem.

fabnuss, 2023-07-28:
hello,

 

So with version 4.13.1 the same piece it was calculated in 3min. and your last version after 30min it had not finished?? so excuse me but there’s really a bug

 
 

Envoyé: samedi 29 juillet 2023 à 00:28

MariMakes, 2023-08-08:
Je m'excuse, je ne parle pas français mais je pense que je peux vous aider.

The shape and description remind me of https://github.com/Ultimaker/Cura/issues/16199, this could be a duplicate. 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-08-23 by github-actions[bot]