Ultimaker/Cura #16273: Floating supports when support set to everywhere

raccooncity-us, 2023-07-25:
### Cura Version

5.4.0

### Operating System

Windows 10 Pro (21H2, Build 19044.3086) 64-bit, Intel Core i7-4790T 

### Printer

Creality Ender 3V2, Klipper, Sprite Pro extruder

### Reproduction steps

1. open STL file with overhangs not touching buildplate
2. enable generate supports with default settings
3. slice file with default settings
4. preview

### Actual results

When panning around the model, you can see the supports are floating ~1mm off the print surface.

### Expected results

Supports should start on the top layer of the print to support above overhangs. I reverted back to Cura 5.2.1 and the supports generated correctly, so this is not an issue with the STL but with 5.4

### Add your .zip and screenshots here ⬇️

[bug_report_floating_supports.zip](https://github.com/Ultimaker/Cura/files/12163680/bug_report_floating_supports.zip)

GregValiant, 2023-07-25:
Thanks for the report.
This looks like a bug to me as well.
Here you can see that although the Bottom Interface is set to .8mm thick with "0" Z gap, it is only 0.20 thick with 0.60 Z gap.  The gcode reflects this as well so it isn't only the display.
[image]

The Cura team will take a look.
@MariMakes this is a new one.

Here I've scaled the model up to 300% and the problem is gone.
[image]

This is the 100% model with tree supports and the problem has gone away.
[image]

raccooncity-us, 2023-07-25:
Glad I could help! I am new to trees so I haven’t played with them yet. I did the same thing as you, and tried adjusting the z gap. I’ll give trees a go for my next print!-Keith

MariMakes, 2023-07-27:
Hey @raccooncity-us,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I showed your issue to the team. They mentioned that it might be related to your Support Interface Settings. 
If I disabled Support Interface Settings the support shows up correctly, without big gaps, without scaling. 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

raccooncity-us, 2023-07-27:
Mariska,

I did some research on the "Support Interface" function of Cura, if I understand it correctly, it creates a roof at the top of the support structure to make the print surface cleaner.

I think this is still a bug for the following reasons:

1) support interface is enabled by default in both 5.2.1 and 5.4, but causes floating issues in 5.4
2) support interface is a published feature of the Curaslicer, so disabling it in this version is only a workaround
2) if I set support interface thickness to 0, the support structure stops floating, but without the enhanced surface contact
3) the "support interface" which contacts the print surface looks different in 5.4 than in 5.2.1, so this section of code has been modified from 5.2 to 5.4.
4) when changing the "support interface thickness" to 0, the supports contact the surface properly (see screenshot). when you set it back to default, you can see that the top and bottom of the support structure change to the enhanced interface surface, but it floats.

If you set "support interface thickness" equal to the layer height it works fine (see screenshot). if you set the layer to .4 for example, you'll see the support structure lifts off the surface by .2 (one layer height), if you set it to .6 it is .4 off the surface (2 layer heights). It almost looks like there is a calculation error in this section of code and you can't customize the thickness of the "support interface" and the default calculation (bound to a function in Cura code) is not outputting a correct value.

<code> floating gap = support interface thickness - layer height </code>

that seems to be the bad math in there. the team should be able to reproduce this easily.

**Cura 5.2.1 - support interface enabled; no floating**
[image]

**Cura 5.4 - support interface enabled; floating; .8 thickness interface to the part (4 layers)**
[image]

**Cura 5.4 - support interface enabled; no floating; 0 thickness, no enhanced surface**
[image]

**Cura 5.4 - support interface enabled; no floating; .2 thickness, .2 thickness interface to the part (1 layer)**
[image]

[CE3E3V2_RR10 Bomber Mamba Mount_support_interface.zip](https://github.com/Ultimaker/Cura/files/12186324/CE3E3V2_RR10.Bomber.Mamba.Mount_support_interface.zip)