Ultimaker/Cura #16276: Clockspring Toaster Slicing Failed

enewbold, 2023-07-26:
### Cura Version

5.4

### Operating System

Windows 11

### Printer

Flsun V400

### Name abnormal settings

none

### Describe model location

Won't slice in any position

### Describe your model

Sliced OK in Orca 1.6.3.  It's a sample file for testing.

### Add your .zip here ⬇️

[FV_Clockspring3D Torture Toaster v24.zip](https://github.com/Ultimaker/Cura/files/12166850/FV_Clockspring3D.Torture.Toaster.v24.zip)
[curalog.zip](https://github.com/Ultimaker/Cura/files/12166855/curalog.zip)

enewbold, 2023-07-26:
Cura 5.4 pretty much is crashing regularly and not slicing much now.  Not a very good situation.

GregValiant, 2023-07-26:
Thanks for the report.
Your right.  I can pretty much copy and paste this response.

There is a persistent bug that keeps good models from slicing.  Sometimes moving the model away from the midpoint of the plate, or rotating the model, or changing the Initial Layer Height but a little bit can fool the bug.
In this case I moved it and rotated it and it sliced.
[image]

I'll mark this as a duplicate and leave it open for the Cura team.  They have a lot of examples, but no toaster that I'm aware of.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic