Ultimaker/Cura #16277: Overlap doesn´t work properly

Junior109007, 2023-07-26:
### Cura Version

5.4

### Operating System

Windows

### Printer

Kingroom KP5L

### Reproduction steps

That is not a bug actually but it is something that do not have on Cura, i notice a really common problem when printing, in Top on Bottom layers the infill does not connect with the wall line, so , i looked for a way to increase the overlap of the infill like i was doing when i used Simplify but do not exist, and part of my print doesn´t work so well because of this.  The "Skin overlap Porcentage" that exist on Cura conect the Inner wall line to the Outter wall line but do not conect the ZIGZAG lines to the Inner line, and what happens is that this diference make defects to the top and bottom surfaces. The bottom and top infill do not connect properly to the walls making holes on the object.

### Actual results

*

### Expected results

*

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-07-26:
Thanks for the report.  A project file would be really helpful.
I agree there is no bug here but there is also no setting to mitigate the situation.  The larger the nozzle, the worse it is and if there is any under-extrusion things get even more noticeable.

There is no setting for that because the Extra Skin Wall is simply a frame for the ZigZags.  If you set the Extra Skin Wall Count to 0 then the ZigZags will be the outer-most feature of the skins and the "Skin Overlap Percentage" will then work directly on the gap between the ZigZags and the Inner-Wall rather than between the Extra Skin Wall and the Inner-Wall.

This is with Extra Skin Wall at 0 and the Skin Overlap at 100% just to visualize.  The ends of the ZigZags are buried in the Inner-Wall.
[image]

This is with the extra skin wall count at 1 and the overlap at 50% to make things easier to see.  You can see that the overlap acts on the distance between Extra Skin Wall and the Inner-Wall.
Therein lies the problem.  So this is expected behavior given the way the software is written.
[image]

Junior109007, 2023-07-26:
Greg, Thank you man. You are a friend.

MariMakes, 2023-08-08:
Hey all 👋 ,

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand.

closed 2023-08-08 by MariMakes