Ultimaker/Cura #16280: Slicing Failed Unknown error

cscottwatson, 2023-07-26:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Ultimaker 2+

### Name abnormal settings

None known

### Describe model location

Centre - does not slice when moved

### Describe your model

Tried mesh fix

### Add your .zip here ⬇️

[UM2_dsm1m_2022_central_london_model2.zip](https://github.com/Ultimaker/Cura/files/12171799/UM2_dsm1m_2022_central_london_model2.zip)

GregValiant, 2023-07-26:
Thanks for the report and the file.
Although the model is watertight it has errors.  This model simply won't slice in 5.4.0.  This really isn't a good model for FDM.
This is in 4.13.1
[image]

This is in MS 3D Builder.  It was no able to repair the model.
[image]

Here it is sliced with PrusaSlicer.  Not good.
[image]

Here is IdeaMaker which has a builtin repair utility that seems to have worked.
[image]

Given the state of the model, I don't know if a bug can be found using this project.  I'll leave the label on and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic