Ultimaker/Cura #16281: Profiles and printers not picked up on update to 5.4.0

N6REJ, 2023-07-26:
### Cura Version

5.4.0

### Operating System

win 11 x64 pro

### Printer

Voxelab Aquila S2

### Reproduction steps

install cura 5.4.0 stable. 

### Actual results

 upon opening you'll be forced to create a new printer and all profiles are gone.

### Expected results

Profiles and printers carry over from 5.3

### Add your .zip and screenshots here ⬇️

none

MariMakes, 2023-07-27:
Hey @N6REJ,

Thanks for your report 👍

You are the first one reporting this 😮 
I've seen it internally if we experiment with configuration-breaking changes, but this shouldn't be the normal use. 

Can you tell us more about your upgrade process?

- What version of Cura did you have before? 
- Did you install the .exe or the .MSI
- What kind of special printer/profiles did you have? 
- What folders show up if you go to %APPDATA%\cura\
[image]
If you still see you previous version there you should be able to restore the settings

N6REJ, 2023-07-27:
5.3.1 is the working cura
5.4 was installed using the .msi
I had a printer profile for my voxelab aqulla s2 and several filament profiles.
[image]
[image]
[image]

MariMakes, 2023-08-09:
Hey @N6REJ,

I'm not sure, but I think what you are seeing might be related to: https://github.com/Ultimaker/Cura/issues/16007

Your screenshots look very normal to me, are these the screenshots from your working 5.3 or your broken 5.4?

N6REJ, 2023-08-10:
5.3