Ultimaker/Cura #16284: Upgrade Cura with the latest understanding of 3D printer physics

kolergy, 2023-07-26:
### Is your feature request related to a problem?

Some of the Cura settings do not represent how the physics of printing actually works and therefore forces the user to modify these settings for different types of prints.

From the speedboat race, notably the results of 247printing and Nitram and in addition with the works of CNC-Kitchen and many others  it is now understood that the maximum print speed depends of only 3 factors:
- The capacity to move to the correct position (capacity to generate a force to accelerate, and capacity of the printer not to deform too much)
- The capacity to melt plastic
- The capacity to cool the melted plastic

The actual physical speed of movement is not a limiting factor, setting the speed is not the causal parameter neither for limitation nor for quality. 

If we review the limitations that affects the speed of a print, there is: 
- The time for the layer to sufficiently cool down before applying the next layer, (already implemented as "minimum layer time")
- The jerk to over simplify representing the limitation on rigidity of the machine. 
- The Acceleration is already implemented but torque of steppers motors depend of rotational speed so it would be good to improve it as acceleration f(speed) 
- the nozzle volume flow-rate is calculated and can be displayed but is missing as a limiting factor and should be added as a limitation

In addition some of the modern printers have very high dynamic capabilities meaning that they are mostly hitting the constraint on their capability to melt plastic.ie: they are volume flow rate limited.

So this is an important parameter to implement knowing that it is already available in Prusa slicer for a while and it is already calculated in Cura as we can display it but not control it.  

I have been waiting for upgrades in this direction since I've seen it appear in Prusa slicer and heard it was in development here however I have recently realized that a change going in that direction had been rejected  see: https://github.com/Ultimaker/Cura/issues/5248 and https://github.com/Ultimaker/CuraEngine/pull/1814, I can not understand for me it feels like rejecting the physics and not admitting that earth is round. 

I implore you to bring Cura to the latest understanding of 3D printer physics.

### Describe the solution you'd like

A slicer that stays up to date with the understanding of the physics. 
Providing consistent prints with different or variable layer heights

Use the nozzle maximum flow rate as a constraint to limit speed  
Use acceleration limitation as a function of speed.

### Describe alternatives you've considered

The alternative is to keep the current settings and continue not being representative of physics and stay in the middle age of 3D printing.

### Affected users and/or printers

All the users who want to use these parameters

### Additional information & file uploads

Some key CNC-Kitchen videos on flow rate: 
- https://www.youtube.com/watch?v=lBi0-NotcP0
- https://www.youtube.com/watch?v=UNJdv5bFGOg

some key 247 Videos: 
- https://www.youtube.com/watch?v=euJYSBsSCzs
- https://www.youtube.com/watch?v=IRUQBTPgon4
- https://www.youtube.com/watch?v=vso6lgTRSo0

speed acceleration curves: 
- https://www.youtube.com/watch?v=MS55OJSeFME

GregValiant, 2023-07-26:
There are a lot of things I don't understand.  This is just another example.
My printer is good up to about 12mm³/sec.  I set the print speed so it doesn't exceed that.  When I print with Silky PLA I am WAY under that as the prints look much better when the outer walls go down slow.
So I don't really understand what advantage there is to this.  The "Flow Rate Compensation Factor" does have some similar effect.  For example - when printing the Infill at 2X layer height the print speed is adjusted downward.
In addition I don't believe UltiMaker printers are constructed to benefit from this.  That is a significant point as UltiMaker does most of the heavy lifting on Cura.  So it would appear that if this was to be added that it would need to come from the community.

ansonl, 2023-07-29:
@GregValiant Flow Rate Compensation Factor as it is currently described isn't clear length of the bowden tube into account for calculating the advance. I searched for uses of it in Cura Engine and it is only used in LayerPlan.cpp https://github.com/Ultimaker/CuraEngine/blob/1c016a0bb618489048b79fba75b8aac7658b484b/src/LayerPlan.cpp#L1773
There aren't any instructions on the relationship between the `extrusion_offset_factor` set by Flow Rate Compensation Factor  in the code and the user's printer setup e.g. bowden or direct , long or short bowden.

mhier, 2023-08-03:
After being pointed here from my duplicate request (sorry again for not searching carefully enough), I would like to second this request about the maximum flow rate. As stated in my (now closed) ticket, there is another aspect justifying this request: Some material is limiting the extrusion rate (=flow) more strongly than others, for instance TPU can be printed usually only a lot slower than PLA due to its softness. This requires to reduce the printing speed not only to a much lower than usual speed, but also the speed will depend on the layer height and nozzle diameter.

The same is true if one is limited by cooling capacity (e.g. when printing ABS, which is a lot hotter than PLA). The amount of heat to be transported away depends to a fairly good extend on the amount of hot material being extruded, so I have to print thicker layers slower, keeping the extrusion rate constant.

Hence, it would be a great help, if Cura could select the speed based upon two speed limits. The currently available speed limit is designed to help with limited spatial accuracy, hence I can select different speed limits for outer walls, infill etc., since the acceptable tolerances are different. Being able to limit the extrusion speed separately would reduce the need for changing these speed settings constantly when switching materials, layer heights and nozzle diameters.

On top of this this will be extra helpful when combining with the Material Settings plugin (https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MaterialSettingsPlugin). I would define the extrusion speed limit for each material while having the normal speed limits in place to ensure my printer moves precise enough when I want it to.

In my opinion, this will be a very useful and convenient addition to Cura.

PS: I have to admit I do not fully understand the description of the "Flow Rate Compensation Factor", but it sounds to me more like a compensation for extruder slippage to avoid under/over extrusion. This is not at all what we are talking about here.

PPS: About the second point regarding the acceleration: In my opinion this would be handled better in the printer firmware, since ideally the acceleration should be reduced gradually during the ongoing move, depending on the momentary speed (which is changing due to the acceleration). This cannot really be done by the slicer.