Ultimaker/Cura #16285: Slicing failed

furmanrobot, 2023-07-26:
### Cura Version

5.30 beta 2

### Operating System

Windows 11

### Printer

Ultimaker 2+

### Name abnormal settings

normal, but with tree supports, bed only

### Describe model location

did not try that.

### Describe your model

Downloaded dragon head. Converted mesh to nurb in Rhino. A warning popped up saying it was not water tight, but I have been getting that a lot lately, so I ignored it. I couldn't see any thing in X Ray mode.

### Add your .zip here ⬇️

[New folder.zip](https://github.com/Ultimaker/Cura/files/12173966/New.folder.zip)

GregValiant, 2023-07-26:
Thanks for the report and the file.
"Not Watertight" usually means that there are seams in the model that aren't connected.  If you tried to fill the model with water it would leak out any open seam.  The problem in 3D printing is that the slicer struggles to figure out what is the inside of the model and what is the outside.
Since you are on Windows you should get MS 3D Builder.  It's pretty good at repairing and altering models.  There are also on-line repair utilities.
I repaired the model with MS 3D Builder (the error seems to have been minor) and this is the repaired model.
[image]

Since the repaired model sliced is this a bug?  I have no idea.  I'll leave this for the Cura team.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic