Ultimaker/Cura #16287: Slicing Adapters

djxs2000, 2023-07-26:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Ender 3

### Describe your model

Garden Hose Adapters

### Add your .zip here ⬇️

[CE3E3V2_Adapter.zip](https://github.com/Ultimaker/Cura/files/12176440/CE3E3V2_Adapter.zip)

GregValiant, 2023-07-26:
Thanks for the report and the file.
It's the nut.  It simply won't slice at .20 Initial Layer Height.  I did get all three to slice at 0.23 Initial Layer Height.
[image]

I'll mark this as a duplicate and the Cura team will take a look.

I print a lot of threaded parts.  You will need to block the support from the threads as it is a real pain trying to clean them out.

djxs2000, 2023-07-26:
Awesome advice, and fast as lightning, thank you so much!

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic