Ultimaker/Cura #16290: 3 walls error but 2 or 4 walls work

PartDaddy, 2023-07-26:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

SeeMeCNC 3MF Project file only

### Reproduction steps

1. open the SeeMeCNC Cura 5.4.0 Project attached to duplicate my exact settings
2. Set walls to 3 and error happens (see photo)
3. Set 2 walls or 4 walls and it works as expected

side not, my STL model is not copyrighted and can be used or shared publicly. The hole cap was quickly drawn in Solidworks 2021.

### Actual results

You should notice slicing results as expected with 2 or 4 walls.  However, 3 walls creates a slicing error.

### Expected results

Should have solid top layer no matter how many walls. 

### Add your .zip and screenshots here ⬇️

Open my project file attached to use my exact profile settings. Thank you for taking the time to review this. 

[SeeMeCNC-Cura540-3wall-error.zip](https://github.com/Ultimaker/Cura/files/12177358/SeeMeCNC-Cura540-3wall-error.zip)

GregValiant, 2023-07-27:
Thanks for the report and the file.
I see you want to close out the infill so it doesn't extend through the skins into those tangs but your Skin Expand Distance of 1.5 is too large and it's interfering with the slice.  The base of the tangs are less than 3mm wide so your setting is causing an overlap of sorts under the tang .  It will work at 1.45 Skin Expand DIstance and the infill will still be short-stopped.

I'll remove the bug label and close this as the setting you are trying to use is inappropriate for the geometry of this particular model..  Let me know if you feel that is premature.

This is with 3 walls and the Skin Expand Distance at 1.45.
[image]

closed 2023-07-27 by GregValiant

PartDaddy, 2023-07-27:
face palm, sorry to bother