Ultimaker/Cura #16292: CE3E3V2_StatueZonai006.3mf

Loumio23, 2023-07-26:
### Cura Version

5.4.0

### Operating System

Windows11

### Printer

Creality Ender 3

### Describe your model

I have sliced many models successfully before, it's the first time. But, it's the first 3d model that I made but the first time I suceed to sliced my model but when I restart cura after the configuration of my 3d printer, the slicing failed

### Add your .zip here ⬇️

[CE3E3V2_StatueZonai006.zip](https://github.com/Ultimaker/Cura/files/12177915/CE3E3V2_StatueZonai006.zip)

GregValiant, 2023-07-27:
Thanks for the report and the file.
After moving it a few times I found a spot on the Cura build plate where it would slice.
I'll mark this as a duplicate report and the Cura team will take a look.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic