Ultimaker/Cura #16293: Print Settings - custom profile

min-hyeok-Park, 2023-07-27:
### Is your feature request related to a problem?

I want to start the program with my pre-set preferences.

### Describe the solution you'd like

I'd like the user to select the desired environment settings and run the program.

However, it is complicated to select all the preferences that exist in Cura, so I made about 4 preferences in advance.

On the web page, the user selects one of these 4 environment settings and clicks 'Start'. At this time, I want the program to run with the selected environment settings.

### Describe alternatives you've considered

I think the contents of the environment settings I set are saved in the following path.
\AppData\Roaming\cura\5.5\quality_changes

So I'm looking for a part of the code that reads and sets this file.

But i couldn't find it. I am asking for your help. I'm asking if there's a better way, or if you can find the code I'm looking for.

### Affected users and/or printers

nothing.

fieldOfView, 2023-07-27:
You can save a project with your custom print settings and an empty buildplate, and open Cura with that project file.

min-hyeok-Park, 2023-07-28:
> 사용자 정의 인쇄 설정 및 빈 빌드 플레이트로 프로젝트를 저장하고 해당 프로젝트 파일로 Cura를 열 수 있습니다.

Thank you for your time.
I don't know if I'm doing well. I'm currently building my own local website to run the cura program on.
The part I want to add here is that when one of the four environment settings is selected on the website, the Cura program is executed based on the environment setting.
The picture below is the configuration I was talking about.
->So..I want my custom environment settings that I selected on the website to be applied before or after running the Cura program.

can you please help me? (You can also say, "Put some code at some point." Or you can suggest a new method.)

[image]