Ultimaker/Cura #16295: Slicing rope bowl fails with 0.15 profile, finishes with 0.2

sanslos, 2023-07-27:
### Cura Version

5.4.0

### Operating System

Windows 10 Pro

### Printer

RadZ (Modified M3D Promega)

### Name abnormal settings

Fails with "Normal 0.15", works with "Draft 0.20"

### Describe model location

Fails in center and off center. Works with supports when rotated 60 deg

### Describe your model

Simple vase-like design with significant detail (rope)

### Add your .zip here ⬇️

Fails with no supports, centered, with "Normal 0.15mm" profile.
Fails with supports, centered, with "Normal 0.15mm" profile.  Hangs with the progress bar at zero (2nd time through, I assume for supports?), saying "Slicing...", 15 minutes have passed and nothing is happening.  Process is not taking any CPU cycles.
Fails with no supports, off center, with "Normal 0.15mm" profile.
Fails with supports, rotated 90 degrees, with "Normal 0.15mm" profile.

Works with no supports, centered, with "Draft 0.2mm" profile.
Works with or without supports, rotated 60 degrees, with "Normal 0.15mm" profile.

[CFFFP_Rope Bowl.zip](https://github.com/Ultimaker/Cura/files/12179344/CFFFP_Rope.Bowl.zip)

GregValiant, 2023-07-27:
Thanks for the report and the file.
The model has errors that are keeping it from slicing.
This is the report from formware.co/onlineSTLrepair.  That site timed out before the model could be repaired.
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> **775 Non-manifold edges** (?)
--> **1143 Inverted faces** (?)
--> 0 Degenerate faces (?)
--> **572 Duplicate faces** (?)
--> 0 Disjoint shells (?)
So the model has a lot of problems. 

MS 3D Builder was able to repair this model.  Here is the repaired model sliced without support.
[image]

This is sliced with normal support "everywhere" at the max of 63° overhang angle.  Slicing takes a while.
[image]

I'm going to remove the bug label and close this.  It appears that it is the model errors that are keeping this from slicing.  I hope you understand.

closed 2023-07-27 by GregValiant

sanslos, 2023-07-27:
Thank you for the information.  I hadn't thought about model errors, I've seen this model printed in various environments which led me to suggest the model was fine.

In the future I will try to review the designs in Builder to check for errors first.