Ultimaker/Cura #16300: Cura 5.4 USB printing timeouts

tabshadel, 2023-07-27:
### Cura Version

5.4.0

### Operating System

Windows 10 professional

### Printer

Creality Ender-5 pro

### Reproduction steps

#D printer connected to Windows 10 PC over a USB Port

### Actual results

On 3 different occasions, printer suddenly stops printing when printing through the USB port

### Expected results

On previous versions of Cura, printing via USB always worked without any problems. So I saved the Gcode file to an SD card and plugged that into the 3D printer and the print worked just fin` when printing from the SD card.`

### Add your .zip and screenshots here ⬇️

Cannot attach log file.

GregValiant, 2023-07-27:
Thanks for the report.
I haven't fooled with this so you will have to experiment.
There are two versions of "USB Printing" included with Cura 5.4.
Go to the MarketPlace and select the "Gear" icon.  Scroll down and if "USB Printing" is enabled then disable it.
Still in the MarketPlace -  click on the "Plugins" button to the left of the Gear button.  Scroll down and Install "USB Printing NG".
The NG version appears to be an update (much needed) to USB printing and the version number is different.  See if you have better luck with the NG version.  Let us know how it works out for you.

I know a lot of people want to print via the USB but if your computer is connected to the internet then Microsoft can push out updates and the computer will re-boot.  That will can a print to fail.
Another thing that happened with the old version is what you experienced.  The print could pause itself and needs a button click to resume.

tabshadel, 2023-07-28:
Thank you for your answers. I will give it a shot and let you know the results.Best regards,Adel Tabsh

MariMakes, 2023-08-10:
Hey @tabshadel,

Did you get a chance to try the suggestions from @GregValiant?
Or are you still stuck?

Would you still like to keep this issue open or do you agree that we can close it?

tabshadel, 2023-08-12:
Hi yes I tried his suggestions and got back the same results. Basically, USB printing from Cura 5.4 does not work.I then got busy learning and instating octoprint which printed FLAWLESSLY through the USB port. I sliced the STL object using cura and then saved the Gcode file and printed it through octoprint.It’s unfortunate that the USB printing got broken in Cura 5.4 and I suggest that some more effort goes towards fixing this issue. I’ll be happy to help test any beta versions that have this feature corrected.Best regards,Adel Tabsh