Ultimaker/Cura #16302: Cura can't start Oops ~~

bbtan89, 2023-07-27:
### Cura Version

5.4.0

### Operating System

window 10

### Printer

Creality 3D ENDER-5

### Reproduction steps

Step 1. Export printing settings from version 5.2.0 that was used previously
step 2. Delete the 5.2.0 version you were using
step 3. 5.4.0 is installed but not executed
Step 4. Installation of 5.2.0 does not work
step 5. Error occurs even when administrator is executed
step 6. Deleting the basic configuration file cura.cfg does not work
Step 7. Backup and reset configuration does not work

### Actual results

Error traceback

Traceback (most recent call last):
  File "cura_app.py", line 239, in <module>
  File "cura\CuraApplication.py", line 829, in run
    super().run()
  File "UM\Qt\QtApplication.py", line 381, in run
    super().run()
  File "UM\Application.py", line 268, in run
    self.startSplashWindowPhase()
  File "cura\CuraApplication.py", line 549, in startSplashWindowPhase
    self._container_registry.loadAllMetadata()
  File "cura\Settings\CuraContainerRegistry.py", line 379, in loadAllMetadata
    super().loadAllMetadata()
  File "UM\Settings\ContainerRegistry.py", line 458, in loadAllMetadata
    cursor = self._getDatabaseConnection().cursor()
  File "UM\Settings\ContainerRegistry.py", line 376, in _getDatabaseConnection
    self._db_connection = self._createDatabaseFile(db_path)
  File "UM\Settings\ContainerRegistry.py", line 354, in _createDatabaseFile
    connection = db.Connection(db_path)
sqlite3.OperationalError: unable to open database file

### Expected results

logs

Thread 0x00001e84 (most recent call first):
  File "UM\Backend\Backend.py", line 169 in _storeStderrToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00000678 (most recent call first):
  File "UM\Backend\Backend.py", line 157 in _storeOutputToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000030b8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000379c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002154 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003af0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000039a8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003a68 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000038c8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000001cc (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Current thread 0x00001dec (most recent call first):
  File "cura\CrashHandler.py", line 397 in _logInfoWidget
  File "cura\CrashHandler.py", line 182 in _createDialog
  File "cura\CrashHandler.py", line 86 in __init__
  File "cura_app.py", line 177 in exceptHook

### Add your .zip and screenshots here ⬇️

[image]
Can't upload screenshot..

MariMakes, 2023-08-11:
Hey @bbtan89,

I showed your issue to the team, they mentioned that you can remove containers.db in your current Cura Cache to resolve the issue. You can find the one for 5.4 by copy-pasting this in your file explorer. 
 %APPDATA%\cura\5.4\cache\

If that doesn't work than you'll might need to delete the containers.db in other Cura versions aswell. You can find those here:
C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cache

Can you let us know if that resolves the issue for you? 
Thank you 🙏

bbtan89, 2023-08-12:
Traceback (most recent call last):
  File "cura_app.py", line 239, in <module>
  File "cura\CuraApplication.py", line 829, in run
    super().run()
  File "UM\Qt\QtApplication.py", line 381, in run
    super().run()
  File "UM\Application.py", line 268, in run
    self.startSplashWindowPhase()
  File "cura\CuraApplication.py", line 549, in startSplashWindowPhase
    self._container_registry.loadAllMetadata()
  File "cura\Settings\CuraContainerRegistry.py", line 379, in loadAllMetadata
    super().loadAllMetadata()
  File "UM\Settings\ContainerRegistry.py", line 458, in loadAllMetadata
    cursor = self._getDatabaseConnection().cursor()
  File "UM\Settings\ContainerRegistry.py", line 376, in _getDatabaseConnection
    self._db_connection = self._createDatabaseFile(db_path)
  File "UM\Settings\ContainerRegistry.py", line 354, in _createDatabaseFile
    connection = db.Connection(db_path)
sqlite3.OperationalError: unable to open database file

Thread 0x00009770 (most recent call first):
  File "UM\Backend\Backend.py", line 169 in _storeStderrToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000ab0c (most recent call first):
  File "UM\Backend\Backend.py", line 157 in _storeOutputToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000c164 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000a194 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000c180 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00004170 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000bf90 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000b12c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000b26c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000bc70 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Current thread 0x0000bc04 (most recent call first):
  File "cura\CrashHandler.py", line 397 in _logInfoWidget
  File "cura\CrashHandler.py", line 182 in _createDialog
  File "cura\CrashHandler.py", line 86 in __init__
  File "cura_app.py", line 177 in exceptHook

MariMakes, 2023-08-15:
Hey @bbtan89,

Thanks for sharing the crash report. 
I'm just not sure what to do with this information. 😅

Can you confirm if you still get the crash if remove containers.db from your cache?

bbtan89, 2023-08-16:
Thanks reply
I tried to remove the containers.db file, but the file doesn't exist C:\Users\\AppData\Roaming\cura<Cura version>\cache.What should I do?

"target": "https://github.com/Ultimaker/Cura/issues/16302#issuecomment-1679091955",
"url": "https://github.com/Ultimaker/Cura/issues/16302#issuecomment-1679091955",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2023-09-07 by github-actions[bot]