Ultimaker/Cura #16307: Object does not slice with "Tree" support structure

FabianKer, 2023-07-28:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Anycubic Kobra Neo

### Name abnormal settings

When changing from "Normal" to "Tree" Support Structure

### Describe model location

Yes (after rotating 90°)

### Describe your model

A miniature D&D figure, slicable with "Normal" support structure and after rotating 90°

### Add your .zip here ⬇️

[character.zip](https://github.com/Ultimaker/Cura/files/12197653/character.zip)

GregValiant, 2023-07-28:
Thanks for the report and the file.
I moved it on the build plate and it sliced.
[image]

I'll mark this as a duplicate and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic