Ultimaker/Cura #16308: no corta

ZDABEO, 2023-07-28:
### Cura Version

5.4.0

### Operating System

windows 11

### Printer

solvo v2

### Describe your model

es un cuadro

### Add your .zip here ⬇️

no corta
[SSL_Ingenious.Tumelo-Esboo.zip](https://github.com/Ultimaker/Cura/files/12197938/SSL_Ingenious.Tumelo-Esboo.zip)

GregValiant, 2023-07-28:
Thanks for the report and the file.
I tried a lot of things and I could not get it to slice.  There are no errors in the model and it's just a little block with 4 holes in it.
The Cura team will take a look.

ZDABEO, 2023-07-31:
Hola me podrías como quitar la gota de cebado me esta dando problemas de atasco en el extrusor y se la quiero desactivar
De antemano gracias

Obtener Outlook para Android<https://aka.ms/AAb9ysg>

GregValiant, 2023-08-01:
When you first posted this I played with that model and couldn't get it to slice at all.
Today I opened it again to check on your "priming drop" and low-and-behold the project sliced with no problem.

Your StartUp Gcode has a "priming line" in it.  It looks like it goes along the front edge of the bed from X5 to X160 at a Y of 10 and at a Z of 0.20.
Your Extruder StartUp Gcode has a line 
"G1 F2000 E93 ;load filament"
and the Extruder End Gcode has:
G1 F2000 E-93 ;long retract for filament removal
Every time there is a tool change in the gcode you are retracting 93mm of filament in one extruder and priming with 93mm of filament in the other extruder.  You can adjust them in the Machine Settings.
Because both of those are from the definition file that was written by Sovol it isn't a problem with UltiMaker Cura as UltiMaker isn't responsible for other manufacture's definition files.

ZDABEO, 2023-08-02:
Oo OK lo checare gracias 😁

Obtener Outlook para Android<https://aka.ms/AAb9ysg>

ZDABEO, 2023-08-04:
Por que cura no me deja entrar en mi cuenta y le pido la recuperación de la clave y nunca manda el mensaje y es esta misma cuenta la qué uso para cura

Obtener Outlook para Android<https://aka.ms/AAb9ysg>