Ultimaker/Cura #16309: Slicing failed

GaryDThompson, 2023-07-28:
### Cura Version

5.4.0

### Operating System

Win 10

### Printer

Creality Ender 3

### Reproduction steps

Scaled stl model to 50mm in the x direction

https://www.thingiverse.com/thing:3585875/files Taylorsville_temple.stl

### Actual results

Slicing engine for Cura has stopped working.

### Expected results

Scaled model would have been printed.

### Add your .zip and screenshots here ⬇️

[CE3E3V2_TAYLORSVILLE_TEMPLE.zip](https://github.com/Ultimaker/Cura/files/12198051/CE3E3V2_TAYLORSVILLE_TEMPLE.zip)

GregValiant, 2023-07-28:
Thanks for the report.
The model has a fair number of errors in it.  After repairing the model in MS 3D Builder it sliced.
[image]

I'll remove the bug label and close this as it appears to be a model problem.

If you have trouble slicing a model in 5.4 you can try rotating, moving it around, or changing the Initial Layer Height but a little bit.  That's a known bug.

closed 2023-07-28 by GregValiant

GaryDThompson, 2023-07-28:
Thanks for looking into this so promptly.

GT

[image]<https://user-images.githubusercontent.com/64202104/256940655-d1cb419a-dc7d-4baf-9213-1b60ed459002.png>