Ultimaker/Cura #16310: Cura prints supports before skirt if 'enable support brim' is enabled

SQL-enwiki, 2023-07-29:
### Cura Version

5.4.0

### Operating System

Garuda Linux (Arch-based), NVidia 3070

### Printer

FLSun Q5, FLSun Super Racer, Ender 3 Pro

### Reproduction steps

Load print on build plate
Select a 5-layer wide skirt
Select 'Generate Supports'
Set up a support brim a few lines wide

### Actual results

Cura printed one of the supports, and then did the skirt, resulting in a print failure.

### Expected results

I expected Cura to draw the skirt first, then do the things inside the skirt

### Add your .zip and screenshots here ⬇️

Support Settings:

[image]

Printing the support first:

[image]

Printing the skirt:
[image]

[cura.log](https://github.com/Ultimaker/Cura/files/12205442/cura.log)
[Project.zip](https://github.com/Ultimaker/Cura/files/12205447/Project.zip)

GregValiant, 2023-07-29:
Thanks for the report.
This is odd.  I get exactly the same results so I can duplicate this with your Super Racer...but not with other printers.
HOWEVER...when I change any printer to "Origin at Center" the problem shows up.
@MariMakes or another member of the Cura team will take a look.

This is an Anycubic printer with "Origin at Center" turned OFF.  This shows the expected behavior of _**Skirt First**_.
[image]

The only change I made here was to turn "Origin at Center" ON.  One of the Support Brims goes down first.
[image]

SQL-enwiki, 2023-07-29:
It's entirely possible that I wasn't switching between the printers correctly.

I just looked again, and it for sure does this on the FLSun Q5 (Another Delta), and the FLSun v400 (another delta). 

You are correct - I tried my ender 3 again, and I'm not seeing it there, my bad, I must have had too much going on last night.

MariMakes, 2023-08-07:
Hey @SQL-enwiki,

Welcome to the UltiMaker Cura Github 🚀 
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/15088

One of our developers has been able to resolve it here: https://github.com/Ultimaker/CuraEngine/pull/1922/files 
So you can expect a fix in the next version of Cura.  It was indeed related to the origin of the printer being in the middle or on the side. 

Great find! 👍

closed 2023-08-07 by MariMakes

Thisismydigitalself, 2023-10-16:
Same here - runnning Cura 5.4.0. supports are printed before skirt.
was this solved in later versions?..

GregValiant, 2023-10-16:
Yes.  This appears to be fixed in 5.5beta.

This is 5.4.0 with Support Brim enabled at 5mm width, and Bed Adhesion set to "None".  With the Support Brim missing the support on the next layer was essentially printing over air.
[image]

This is 5.5beta with the same model and same settings.  Looks fixed to me.
[image]

DK4YA, 2024-01-17:
I print with a delta and a cXY. The problem that the support is printed before the skirt occurs with the Delta, but not with the cXY. Everything under Cura 5.4.