Ultimaker/Cura #16312: Gridfinity base slicing fail on 5X5

jwm31337, 2023-07-29:
### Cura Version

5.4

### Operating System

MacOS Big Sur 11.7.8

### Printer

Ender 3 S1 Pro

### Name abnormal settings

No

### Describe model location

It fails even when moving model or rotating

### Describe your model

Only smaller frames (3X4, 4X5, etc)

### Add your .zip here ⬇️

[CE3S1PRO_Frame 5x5.3mf.zip](https://github.com/Ultimaker/Cura/files/12206082/CE3S1PRO_Frame.5x5.3mf.zip)

GregValiant, 2023-07-29:
Thanks for the report and the file.
The model is 210 x 210 and your skirt distance is 10mm all around and the skirt itself is 3mm wide all around.  So 210 + 26 is 236mm which doesn't fit on the build plate.
Change the "Skirt Line Count" to 1 and the "Skirt Distance" to 2mm.  That will make a footprint of 216 which fits the build plate and the model will slice.
This is expected behavior so I'll remove the bug label and close this.

Cura used to put a gray border around the build plate that would indicate the width of the skirt/brim so the user would know what the problem is.  This version of Cura can compensate for that if only pieces of the skirt/brim will fit and so the gray area is gone.  What it can't compensate for is when the _ENTIRE_ skirt/brim won't fit and that is the problem you have here.

In the image below I've moved the model so it is close to the edge of the build plate and the skirt distance is 10 with skirt line at 3.  This slices because at least some of the skirt can be placed on the bed 
[image]

closed 2023-07-29 by GregValiant

jwm31337, 2023-07-29:
> DuckDuckGo removed 1 tracker.
> More →

Greg,
Thanks SO much for your quick response. I’m new to the 3D printing world so I appreciate your help in educating me.

Take care,
jay

Jay Mackey
408.306.2234