Ultimaker/Cura #16315: Bug in Mac cura 5.3.1

1Junglist, 2023-07-29:
### Cura Version

5.3.1

### Operating System

MacOS

### Printer

Elegoo Neptune 3 Pro

### Reproduction steps

1. Slice file
2. Go to preview Sliced item
3. Cura Crashes, unable to select anything have to turn off the macbook pro m1 by power button to restart computer, but works fine on windows system was able to slice and view on windows 10
[image]
[image]
[image]

### Actual results

cura crashes

### Expected results

Should be able to preview sliced file

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]

MariMakes, 2023-08-10:
Hey @1Junglist,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I've never seen a crash like this 😨 
It seems like the preview is not working correctly. 
Are you only seeing this in this model, or also in other models? 

As a Windoows user this issue is outside my field of expertise 😖, so I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞

1Junglist, 2023-08-10:
Hi ya mate,

It only happens on my Macbook pro m1, I have seen it a couple times on the
3D printing Facebook groups.
I think it's software based doesn't show in prints as I can't save the
g.code, the glitch just locks up my whole Macbook and I have to do a hard
reset on the laptop.

Thank you

Ben

MariMakes, 2023-08-11:
Hey @1Junglist,

I showed your issue to the team and they mentioned that it might be related to your hardware. 
Unfortunately, that is not something that we can fix in Cura 😞 

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-08-25 by github-actions[bot]