Ultimaker/Cura #16317: Support interface function supports properly Extruder1 only

zolee1209, 2023-07-29:
### Cura Version

5.4.0

### Operating System

Windows 10 64bit

### Printer

Custom FFF

### Reproduction steps

Slicer does not create support interface layer when the support interface extruder is Extruder2. Or at least, not Extruder1.
It does not matter, what are the other extruder settings in this Support class.
As far as i tested, this function works fine in the Cura 4.13.1 and not from 5.x 

### Actual results

There is no possibility to print the model and support with Extruder1 and the support interface with Extruder2.

### Expected results

As above in the actual results.

### Add your .zip and screenshots here ⬇️

[_support_interface_test.zip](https://github.com/Ultimaker/Cura/files/12207323/_support_interface_test.zip)
[image]
[image]
[image]
[image]
[image]
[image]

GregValiant, 2023-07-29:
Thanks for the report.
It doesn't work for your JG Artist printer because the "Enable Support Interface" checkbox is grayed out and can't be enabled. 
 It does work for other dual extruder printers.
This is in 5.4.0 with a dual extruder Ender.
[image]

This is with a Sovol SV02
[image]

This is with your printer.  Notice that the Enable Support Interface is grayed out and not available.
[image]

If you used a Custom FFF printer the fdmprinter.def.json file says this about "support_interface_enable":
"enabled": "support_enable or support_meshes_present"
Both of those are true in your project so the setting should be enabled.  Which extruder is assigned to Support Interface isn't involved.

Is there some printer definition file or other file that might have overridden that setting?

zolee1209, 2023-08-14:
> Thanks for the report. It doesn't work for your JG Artist printer because the "Enable Support Interface" checkbox is grayed out and can't be enabled. It does work for other dual extruder printers. [...]

Thanks for the reply and sorry for the late response from my side.

The Enable Support Interface checkbox was always inactive on my side, even in the 4.x versions, but worked fine.
I do not have any additional printer definition file or some like this or at least i do not know about it.
I made this printer configuration in an earlier version of Cura and i just imported it with a *.3mf file.
I will try to create a new printer in the current version and see if it behaves differently.

GregValiant, 2023-08-14:
I still haven't figured out why this happened but the two children settings "Enable Support Roof" and "Enable Support Floor" have come un-linked from the parent setting.  It seems to have happened because of manual changes to the "Fine" profile.

To the right of "Fine" in the profile selection box is a circle with an arrow on it.  It's the reset button.  Click on that and "Enable Support Interface" will work as it should.
You will have to go back and make manual changes to the profile (enable the interface, set the interface to Extruder 2, etc) but it worked for me.

zolee1209, 2023-08-14:
I made some tests...
At first, i made a new printer with the same settings, but that didn't solve the problem. The "Enable Support Interface" checkbox remained unactivated.
My next step was that i resetted the Print settings profile, and after that, the checkbox can be activated.
I copied manually all settings of both extruder from the "bad" settings, but the checkbox remained activateable.
I also noticed that although the checkbox is inactive, when displaying the other parameters, the system takes it as activated in the "wrong" configuration.

I'm just guessing that some imported settings are corrupting it, because after I reset the profile and set the old parameters, it works properly.

zolee1209, 2023-08-14:
> I still haven't figured out why this happened but the two children settings "Enable Support Roof" and "Enable Support Floor" have come un-linked from the parent setting. It seems to have happened [...]

Yes, that's exactly what I did and it solved the problem. ;)
Thanks for the help!

closed 2023-08-17 by zolee1209