Ultimaker/Cura #16321: Problem with Cura

foxyjamp, 2023-07-29:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Anycubic Viper

### Name abnormal settings

The last thing I did was put a support blocker on one arm only, because of that I had problems

### Describe model location

No

### Describe your model

No

### Add your .zip here ⬇️

[Error.zip](https://github.com/Ultimaker/Cura/files/12208165/Error.zip)

GregValiant, 2023-07-30:
Thanks for the report.
I can't get it to slice either.  I'll mark this as a duplicate and the Cura team will look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic