Ultimaker/Cura #16322: slicing failed

wluckie1, 2023-07-30:
### Cura Version

5.4.0

### Operating System

windows 11

### Printer

anycubic Kobra 2

### Name abnormal settings

no. i tried slicing with my settings which have been working on other items

### Describe model location

centre of plate. i tried rotating but still doesn't slice

### Describe your model

no i haven't. i tried mesh fix and all seems fine. mesh fix says item is watertight

### Add your .zip here ⬇️

[walther_ppk_toystar_stand.zip](https://github.com/Ultimaker/Cura/files/12208372/walther_ppk_toystar_stand.zip)

GregValiant, 2023-07-30:
Thanks for the report and the file.
I didn't have any trouble slicing this.  Have you tried moving, rotating and/or changing the Initial Layer Height?

wluckie1, 2023-07-30:
Yes I tried rotating it. I also tried slicing it with other profiles with different layer heights. I also tried slicing it with my Vyper profiles for my other printer 
    	Get Outlook for iOS

j-snijder, 2023-07-30:
Same here.
slicing with tree support will not work at all, I guess it has to do with the tree support option becaus with normal support it slices

j-snijder, 2023-07-30:
> 

Everytime I read you devs not having the same issue, so I conclude we users are too dumb to use Cura ? or you devs have some files or settings on your machiines/computers that is upped when updating the repo so normal users get in trouble !

Another explaination is hard to find IMHO

GregValiant, 2023-07-30:
I'm an old retired guy living in Florida.  I don't work for UltiMaker and I'm not a "Dev".  I just use the software and don't have anything better to do so I'm the greeter here.  I've learned a lot doing this.
I have the absolute cheapest laptop I could find and it runs Windows 10 Home.  The video system is an onboard Intel set.

There does seem to be a lot of bug reports here on models that won't slice.  I have found that on about 3/4 of those I can get to slice after fooling with them.  That's a bug in my book.  Error free models should slice.
Now if the model does have errors, that's something different.
I still use 4.13.1 a lot and the SmartAvionics fork for Spiralize.

j-snijder, 2023-08-04:
Today I discovered slicing will not work on a fully new and bare installed windows 11 computer (my old one crashed major and had to buy me a new one)
In my book that means the bug is bigger then can be imagined and my advise would be to roll all updates back to the 4.13 version and start from scratch again.
As its a complete waste of time trying to fix whats unfixable.

Jan-Snijder, 2023-12-24:
SOLVED ! damaged ? file in configuration files.