Ultimaker/Cura #16324: auto scale by 10,000%

LegoDwarf120, 2023-07-30:
### Cura Version

5.4.0

### Operating System

windows 11

### Printer

creality

### Reproduction steps

edited a file on 3d builder
then put the file into cura slicer

### Actual results

it just auto scaled by 10,000%

### Expected results

place normally into slicer and not auto scale

### Add your .zip and screenshots here ⬇️

[image]

flame4ever, 2024-09-09:
I have the same problem, but also with STL files.
[image]

GregValiant, 2024-09-09:
Cura doesn't just make up numbers, it gets them from the model file.  If you know the height, or width, or length, you can adjust the scale.

nallath, 2024-09-09:
This option is there if cura detects _reaaaaaly_ small objects. That's also where the number of 10k comes from, as it tries to figure out what you meant. 

You can disable this in the preferences (i think it's called "automatically scale small objects")

flame4ever, 2024-09-12:
The Prusa Silcer also copes with this and as can be seen in the video, it first has the object at the right size and then makes it larger, can't this be prevented?

flame4ever, 2024-09-16:
@fieldOfView You're right, I set it to mm and the behaviour is gone, thank you very much!!!
[image]

GregValiant, 2024-09-16:
@flame4ever if this is resolved can we close this?