Ultimaker/Cura #16325: Slicing Failed

Experte9, 2023-07-30:
### Cura Version

5.3.1

### Operating System

Windows 10

### Printer

Creality Ender-3-S1

### Name abnormal settings

Sliced 5 oder Models before

### Describe model location

Yes

### Describe your model

Caketopper

### Add your .zip here ⬇️

[Luna_W.zip](https://github.com/Ultimaker/Cura/files/12209000/Luna_W.zip)

GregValiant, 2023-07-30:
Thanks for the report and the file.
This should slice and I didn't have any trouble with it.  I checked the model and it doesn't have any errors.
There is an often-reported bug that can cause this though.  Try moving the model around in Cura.  Sometimes rotating it or changing the Initial Layer Height by a little bit can fool the bug.

I'll mark this as a duplicate and the Cura team will take a look.

Experte9, 2023-07-30:
Okay, I try to move it around and it worked. 
Slicing is done ✅ 
Tanks!

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic