Ultimaker/Cura #16326: Mirror Slicing Error

LordGalfor, 2023-07-30:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Ender Pro 3

### Name abnormal settings

Mirroring File causes slice error

### Describe model location

Fails to slice at all angles

### Describe your model

Slices successfully until only change being the mirror of the file.

### Add your .zip here ⬇️

[CE3PRO_Easefare_BrokenCastleWall.zip](https://github.com/Ultimaker/Cura/files/12209527/CE3PRO_Easefare_BrokenCastleWall.zip)

Slices fine as is. Click "Mirror", and the slicing will fail. Looking for any troublshooting tips!

GregValiant, 2023-07-30:
Thanks for the report and the file.
This bug is truly annoying.  The usual workaround is to move the model away from the midpoint of the build plate, rotate the model, and/or change the Initial Layer Height.
For this model I could only slice the mirrored part when it was AT the midpoint of the build plate.
I'll mark this as a duplicate and the Cura team will take a look.

Here they are - 2 at once.  It took me 15 minutes to figure this one out.  That should not be necessary.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic