Ultimaker/Cura #16328: bug: slicing failed on marble track

HarikrishnanBalagopal, 2023-07-30:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

AnyCubic Kobra Neo

### Name abnormal settings

Experimental: Slicing tolerance: Exclusive

### Describe model location

model location is center (default)

### Describe your model

A complicated marble track with several overhangs

### Add your .zip here ⬇️

[AK_Track.zip](https://github.com/Ultimaker/Cura/files/12210159/AK_Track.zip)
The model is from here https://www.thingiverse.com/thing:1385312

Note: The issue is resolved after changing Experimental: Slicing tolerance to Middle (default)

GregValiant, 2023-08-01:
Thanks for the report and thanks for letting us know it is resolved for you.

This looks like a duplicate of many other bug reports regarding good models that won't slice in 5.4.0.  Your project will slice in both "Middle" and "Inclusive" but to get it to slice "Exclusive" I had to change the Initial Layer Height to 0.23.  It sliced with that single adjustment.
Adjusting the Initial Layer Height also happens to be one of the tricks to fooling that "often reported" bug.

I'll mark this as a duplicate and the Cura team will take a look.  It's possible that the "Exclusive" setting isn't involved other than moving the slice to a Z that Cura doesn't like.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic