Ultimaker/Cura #16334: Material profile import file extension

Asiak3, 2023-07-31:
### Cura Version

5.3.1

### Operating System

Windows 11

### Printer

Velleman K8400

### Reproduction steps

Go to Material Profiles and click on import

### Actual results

The file extension is wrong and dont allow you to import any profile.
It shows:

Ultimaker Material Profile (*.xml.fdm_material)

It schould be:

Ultimaker Material Profile (*.xml *.fdm_material)

### Expected results

Allow you to select the profile to import.

### Add your .zip and screenshots here ⬇️

0

GregValiant, 2023-08-01:
Thanks for the report.
From what I can see all the files in "C:\Program Files\UltiMaker Cura 5.3.1\share\cura\resources\materials" use this naming convention:
3D-Fuel_PLA_PRO_Black **.xml.fdm_material**
and this one is from 5.4.0
3D-Fuel_PLA_PRO_Black **.xml.fdm_material**
The files in 4.13.1 are named in the same manner.

Here is the Import Dialog "Files of type" selection box and it appears to match those file naming conventions:
[image]

Where did you find/obtain the Material profiles you are trying to import?

Asiak3, 2023-08-01:
Thanks for the answer. The correct file format is .xml.fdm_material?
I've got my material from https://www.extrudr.com/en/3d-printing/material-profiles/
If the extension in cura is correct e extrudr gives bad files.
Sorry

fieldOfView, 2023-08-01:
Cura is correct, Extrudr is not. You may also find that the profiles provided by them only includes support for printers that use 2.85 mm material. They do not apply to your printer.

I'm closing this issue because it is not something that Cura is doing wrong.

closed 2023-08-01 by fieldOfView