Ultimaker/Cura #16335: Linux AppImage segmentation faul

CocolinoFan, 2023-07-31:
### Cura Version

5.4.0

### Operating System

Gentoo Linux x86_64 | Kernel: 6.3.9-gentoo-dist 

### Printer

Creality Ender 3 v2

### Reproduction steps

1. `./UltiMaker-Cura-5.4.0-linux.AppImage`
2.  Navigate to: File -> Open File(s) ...

### Actual results

Application crashes and outputs the following:
`/tmp/.mount_UltiMaj92ceU/AppRun: line 25: 12672 Segmentation fault      $scriptdir/UltiMaker-Cura "$@"`

### Expected results

Opening a menu to allow me to select a file to import.

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/12221924/cura.log)

anno0, 2023-07-31:
Maybe this could be the reason

`2023-07-31 21:26:57,586 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [444]: file:///tmp/.mount_UltiMakh81u1/share/cura/resources/qml/Cura.qml:624:5: QML FileDialog: Failed to load non-native FileDialog implementation:
qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml:4 module "Qt.labs.folderlistmodel" is not installed`

ntguardian, 2023-08-29:
Exact same issue on Arch Linux. Can confirm though that the drag-and-drop feature still works. That is an effective work-around for me.

anno0, 2023-08-29:
> Exact same issue on Arch Linux. Can confirm though that the drag-and-drop feature still works. That is an effective work-around for me.

Well, that was my first thought and I was a little bit surprised. :-) But this won't work.

The next bug appears at the"Save on Disk" button. Same bug ...

`(UltiMaker-Cura:9217): Gtk-WARNING **: 18:37:36.631: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Yaru/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Yaru/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
/tmp/.mount_UltiMaNJh3Ab/AppRun: line 25:  9217 Aborted                 (core dumped) $scriptdir/UltiMaker-Cura "$@"`

malnoxon, 2023-09-16:
Same issue, same cura version 5.4.0, same workaround (opening the .stl directly from my file manager through cura works as well). I'm on arch linux. (cura-modern-appimage aur package version 5.4.0-1)

Save to disk works successfully for me.

Issue occurs on trying to import a print settings profile as well. (`/tmp/.mount_curaWVXp3l/AppRun: line 25: 181012 Segmentation fault      (core dumped) $scriptdir/UltiMaker-Cura "$@"`)

kamikazestar, 2023-10-01:
I noticed the same issue while opening the file STL file. The printer, Cura version, and reproduction steps are the same.
Operating system: Manjaro Linux x86_64
Kernel: 6.1.53-1-MANJARO
Actual result: ```/tmp/.mount_UltiMaTxgAmy/AppRun: line 25: 1030183 Segmentation fault      (core dumped) $scriptdir/UltiMaker-Cura "$@"```

For test purposes, I also installed the AUR package ```cura-bin``` and ```cura-5-bin```, and in both cases, I had very similar issues:
- Error from ```cura-bin``` package: ```/usr/bin/cura: line 25: 2585098 Segmentation fault      (core dumped) $scriptdir/UltiMaker-Cura "$@"```
- Error from ```cura-5-bin``` package:  ```/usr/bin/cura5: line 5: 4093128 Segmentation fault      (core dumped) LD_PRELOAD=/usr/lib/libstdc++.so.6 QT_QPA_PLATFORMTHEME=xdgdesktopportal QT_QPA_PLATAFORM=xdgdesktopportal ./UltiMaker-Cura $@```

sergomm, 2023-10-12:
> Same issue, same cura version 5.4.0, same workaround (opening the .stl directly from my file manager through cura works as well). I'm on arch linux. (cura-modern-appimage aur package version [...]

the same behavior on the last Debian and Ubuntu
the same workaround for "Open file" via the command line, but the same workaround absence for "Import profile"((

geraldnilles, 2023-10-16:
I found a workaround that works on my system (Arch linux, Wayland, Sway WM). Figured i'd drop it here to help anyone who is having problems.  I also figure it'll give dev's a hint into the root caus

`./UltiMaker-Cura-5.4.0-linux-modern.AppImage -platformtheme gtk3`

MariMakes, 2023-10-19:
👋 Quick update on our side. 

If your issue is related to xdg-desktop-portal-gtk, than it should be resolved in the 5.5 Beta release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)

If you are experiencing a crash similar as this one related to a missing icon, perhaps it looks like this:
```
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Papirus-Dark/16x16/actions/image-missing.svg: Unable to load image-loading module: libpixbufloader-svg.so: libpixbufloader-svg.so: cannot open shared object file: No such file or directory (gdk-pixbuf-error-quark, 5) Bail out! 
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Papirus-Dark/16x16/actions/image-missing.svg: Unable to load image-loading module: libpixbufloader-svg.so: libpixbufloader-svg.so: cannot open shared object file: No such file or directory (gdk-pixbuf-error-quark, 5)
```
We are collecting cases in this ticket: https://github.com/Ultimaker/Cura/issues/16932
This ticket also contains an Alpha build that perhaps might resolve the crash for you, but it doesn't seem to help everyone. 

Thanks for all the feedback. Together we can make Cura better 😊

closed 2023-10-19 by MariMakes