Ultimaker/Cura #16339: Slicing failed

sparky-corona, 2023-08-01:
### Cura Version

5.3.1

### Operating System

Ubuntu 20.04.03 LTS

### Printer

FLSUN SR

### Name abnormal settings

increased size of print

### Describe model location

center

### Describe your model

yes

### Add your .zip here ⬇️

[holder_v2.zip](https://github.com/Ultimaker/Cura/files/12225957/holder_v2.zip)

GregValiant, 2023-08-01:
Thanks for the report and the file.
One of the Linux users will need to take a look.  I didn't have any trouble with it on Windows.
There is a minor error in the model that could use repair, but it sliced for me without fixing it.
Have you tried moving, rotating the model or changing the Initial Layer Height by a little bit?
[image]

sparky-corona, 2023-08-03:
Can you show me this minor model error you are talking about. I pretty new at this modelling tools and was trying to avoid such errors, but maybe I've just overseen this one.

I tried rotation and sone small shifts from the center. Didn't work.

I also rescaled the model to half the size. This was working andthe print looked fine.

sparky-corona, 2023-08-03:
Today I tried again after restart of Cura an my computer.

Confusingly, now the slicing worked fine, although it took nearly 5min to slice for this model and the model was still in the center of the build plate. Well, anyway. The output file is around 96MB big.

This maybe takes some time, even on a AMD 24-Core Ryzen machine.

Thanks, ether way.

Cheers!

GregValiant, 2023-08-03:
This is in Mesh Mixer
[image]

This is the report from formware.co/OnLineSTLRepair.  This actually isn't too bad.  The "Non-manifold edges" can be a show stopper in Cura. 
-> Analysed your file:
--> 14 Naked edges (?)
--> 3 Planar holes (?)
--> 0 Non-planar holes (?)
--> 1 Non-manifold edges (?)
--> 1 Inverted faces (?)
--> 154 Degenerate faces (?)
--> 2 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 3473ms ------
-> Vertex count changed from 4879 to 5486 (+607)
-> Triangle count changed from 9982 to 11076 (+1094)

sparky-corona, 2023-08-04:
I'm not sure how to perform this check in blender. Do you know that too, maybe?

What are Naked Edges and planar holes, degenerates faces, non-manifold edges and inverted edges?

I was always trying to recalculate the surface normals before combining 2 meshes together or derive the difference between two.

Is there a way I can search for such errors in blender?

Thanks for the info on the found errors/ warnings.