Ultimaker/Cura #16340: HueForge File Failing to Slice

Meph88, 2023-08-01:
### Cura Version

5.4

### Operating System

macOS Ventura 13.4.1

### Printer

Elegoo Neptune 3 (Modded build plate size, used for Neptune 4)

### Name abnormal settings

layer height of 0.08mm

### Describe model location

No

### Describe your model

Slices if I change the scale, but not at the original size or close to it.

### Add your .zip here ⬇️

[CE3E3V2_Samuri_Girl_Front_200x200.3mf.zip](https://github.com/Ultimaker/Cura/files/12225979/CE3E3V2_Samuri_Girl_Front_200x200.3mf.zip)

GregValiant, 2023-08-01:
Thanks for the report and the file.
I can't reproduce the problem but I'm on Windows and it might be a MAC thing.

Cura 5.3 and 5.4 have developed problems when slicing models that have a lot of detail within a couple of layers.  Models like this one (and other litho's), castles with stone floors, cabins with textured wood floors, that sort of thing.  Occasionally, changing the Initial Layer Height by just a bit can fool the bug.

I'll leave this for the Cura team and they will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic