Ultimaker/Cura #16341: I can't install plugins

Manekiss, 2023-08-01:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Creality Ender 3 v2 Neo

### Reproduction steps

1. go to marketplace
2. try to install any plugin

### Actual results

nothing, It just stays charging for a while and then it doesn't anymore.
[image]

if i try to install them from the page and then sync cura an error occurs
[image]

[image]

[image]

### Expected results

be able to install a plugin

### Add your .zip and screenshots here ⬇️

[image]

[image]

[image]

GregValiant, 2023-08-01:
Thanks for the report.
This sounds like a general problem but @5axes is the developer of a couple of those plugins.  Maybe he has run into this problem previously??

5axes, 2023-08-01:
No absolutly no idea or clue

GregValiant, 2023-08-01:
No one else has reported this problem.  I wonder if your anti-virus is getting in the way???

arismelachroinos, 2023-09-03:
I have the same issue.

arismelachroinos, 2023-09-03:
I installed a VPN and it seemed to fix the issue.