Ultimaker/Cura #16342: Extraneous Extrusions in screw thread receptacle

Ergonomicmike, 2023-08-01:
### Cura Version

5.4.0

### Operating System

Win 10

### Printer

Ender 3 v2, mriscoc firmware (latest)

### Reproduction steps

I made a model of a receptacle for a screw using TinkerCAD, I ran it through an on-line STL repair tool. (See project file.) Sliced it. It looks fine in Preveiw.

### Actual results

Occasionally there are straight extrusions across the bottom of the threads. (Photo)

It could be that they are "failures" from the overhang. Except that they're so straight. Not the usual curly strings hanging from overhangs. And one is especially long/egregious that I don't think it's an overhand failure.

There seems to be a pattern to their appearance, in that always on bottom side of thread.  Always about the same length of extrusion. Possibly a geometric pattern where they occur. 

I thought I read that 5.4.0 sometimes did this. (?)

So I thought I should report it as a "Sanity Check." (Is it "Just Me?")

### Expected results

It should look like the preview. No extraneous extrusions.

### Add your .zip and screenshots here ⬇️

[image]

[Pill Crusher Female_for Github.zip](https://github.com/Ultimaker/Cura/files/12227418/Pill.Crusher.Female_for.Github.zip)

GregValiant, 2023-08-01:
Thanks for the report and the file.
I print a lot of threads.  I recognize what is going on and it is likely a settings and/or printer issue.

The crux of the problem is that the extrusions aren't staying where they're put.  The tool path is correct, but the extrusion wants to pull in a straight line behind the nozzle rather than staying on the path.  This can be a "layer adhesion" (temperature) problem, a "flow" problem, or a mechanical problem produced by your settings (walls).

- You are printing at 201.  I would suggest turning that up a bit to aid the layer adhesion.  At 210 you would still be well within the PLA ballpark.
- Looking at the flat area in the bottom of the cavity - I can see between the strands of the top layer.  That's under-extruded by at .east 10% and is a definite problem when getting extrusions to stay where they belong.  You have changed your firmware - did you calibrate the E-steps after doing that?  Flat surfaces are the best for judging Flow.  Use a decent magnifying glass and look at the surface and you will see what I mean.  All the extrusions on a skin should be welded together with no gaps anywhere.
- Turn off coasting.  It's OK for external show surfaces to hide the Zseam a bit better, but it works by under-extruding the last couple of mm's of a run and you don't need anymore under-extrusion.
- Consider printing with at least 3 walls.  The threads could use the extra strength.  I often drop in a support blocker for the threads and configure it as an Infill Mesh with additional walls.  The leaves the non-critical exterior to print faster.

I don't see a bug here but rather a combination of other factors so I'm going to remove the bug label.  I'll leave this open so you can respond.  I hope you understand.
-

Ergonomicmike, 2023-08-01:
Thanks for the quick reply. I can accept that it's a print problem at my end. 

> I often drop in a support blocker for the threads and configure it as an Infill Mesh with additional walls.

I haven't heard about a support blocker. Can you give me a link, please?

As for temps, E-steps, extrusion, etc., AFAIK all perfect here. At least according to Orca slicer's Extrusion test. But I see too that there are gaps in that layer. (I'll increase the flow to see what happens.

Temps are ironic. Higher temp = better adhesion. But also higer temp = more droop on overhangs. But I'll try higher.

I'll close this issue shortly.

GregValiant, 2023-08-01:
What I've done here is brought in a "Cylindric Custom Support".  The plugin is available in the MarketPlace.
I scale the cylinder and move it into position with the tapped hold.
In the "Per Model Settings" I set it to "Modify Overlaps" and as an "Infill Mesh".
The only setting I am using for this is "Wall Count" so the others caan be dismissed.  I have set the "wall count" to 5.
I have adjusted a couple of the regular Cura settings to make this easier to see (no infill and a single wall).  Now the area of the threads is a cylinder that looks like it's been tapped.  This is a lot more robust.
[image]

This is without that extra cylinder.  
[image]

Ergonomicmike, 2023-08-01:
Thanks very much. Above and beyond the call of duty.

closed 2023-08-01 by Ergonomicmike

Ergonomicmike, 2023-10-21:
@GregValiant I wanted to thank you again for your help/suggestion about how to print threads.

I had a store-bought pill crusher that failed. I couldn't easily find a replacement. So I decided to print my own. (I can't swallow pills anymore.)

As typical with me, it took a few iterations. (I had to use your trick also on the male part to reinforce the center of the top of the crusher to keep it from flexing under load.) But I now have a printed pill-crusher that works, thanks to you.
[image]

While other slicers might slice better than Cura at this point in time, one thing that differentiates Cura from the rest is the plugins. I couldn't have done with without @5axes 's Cylindrical Custom Support.

GregValiant, 2023-10-21:
When I design threads now I use my own thread form that uses a 70° included angle rather than the standard 60°.  The threads themselves don't require support and are more forgiving in regards to tolerances between the male and the female.

If you happen to need a screw connection that same trick with the cylinder will work to increase the wall count around the through hole.  That keeps it from collapsing/deforming when the screw and nut fasteners are tightened.
PLA in particular will still be subject to cold flow deformation that will cause the screw connection to appear to loosen, but PETG holds up pretty well.

Ergonomicmike, 2023-10-22:
Thanks. When/if it fails, I'll plan to print it in PETG.

As for the 70 degree angle, I'll have to cheat and use Cura to scale the threads in Z to increase the angle a bit more.