Ultimaker/Cura #16343: Bitmap post-processing script doesn't "show" anything when filament color is Black

Ergonomicmike, 2023-08-01:
### Cura Version

5.4.0

### Operating System

Win 10

### Printer

Ender 3 v2

### Reproduction steps

I changed a filament color to black. Suddenly the bitmap preview stopped "working" on my printer's display.

### Actual results

Well, actually, it was working. But nothing was showing on my display because black filament against a black background shows all black! (It took me a few hours to figure this one out. I reformed my SD Card, thinking it might be a corrupt bit on the card.)

### Expected results

Prints that are printed in black filament should be viewable in the bitmap image.

I suggest that when a user has set the color of the filament to black that the background of the bitmap preview change to white. (A.k.a, Reverse Video.)

### Add your .zip and screenshots here ⬇️

none

GregValiant, 2023-08-01:
Thanks for the report.
Is there something wrong with the bitmap or is it just that you can't see it because of the primary color?
Are you using the "CreateThumbnail" post processor to generate the image?

Ergonomicmike, 2023-08-01:
Apparently I wasn't clear. It's black on black. Nothing shows in the bitmap when you select black color filament.

GregValiant, 2023-08-01:
OK.  Not a bug then as the system is working, just no contrast.  The reason I had asked was to find out if something was broken, or if it is simply hard to see.
I don't think a bug report is proper for this.  It would get better traction as a feature request.  The plugin that provides the picture just takes a snapshot of the screen and converts it to a text description.  There is nothing in the code to actually "read" the image or to adjust the image.  The whole plugin is only 109 lines long.

I'll go ahead and close this and you can open a feature request to broaden the scope of the "CreateThumbnail" plugin.

closed 2023-08-01 by GregValiant