Ultimaker/Cura #16344: Plush Octopus

rpoliquin, 2023-08-01:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Sovol SV04

### Describe your model

I have tried slicing this with various settings. It appears to be an issue with the layer height. A layer height of 0.2mm slices, but all other options fail

### Add your .zip here ⬇️

[S1_Octopus_Plush_Angry_by_BODY3D.zip](https://github.com/Ultimaker/Cura/files/12228316/S1_Octopus_Plush_Angry_by_BODY3D.zip)
I have been able to slice a slightly different version of this model with a .15mm layer height, but for some reason this model will only slice with 0.2mm layer height

GregValiant, 2023-08-01:
Thanks for the report and the file.
There is a persistent bug in 5.3.1 and now 5.4.0 that can keep a model from slicing.  There are three workarounds that many times can fool the bug.
Moving or rotating the model on the Cura build plate are two of them.
The third is changing the Initial Layer Height.
Since you were OK at .20 layer height I thought the third workaround had a chance so I changed the Initial Layer Height to .21 and the model sliced with the Layer Height at .15.

I'll mark this as a duplicate slicing error.  Since the Cura team has numerous examples of this behavior I'll go ahead and close this.  I hope you understand.

rpoliquin, 2023-08-01:
Thank you so much for the recommendations. I’ll try those fixes next time I
see a problem.