Ultimaker/Cura #16345: wont slice

natennd, 2023-08-01:
### Cura Version

5.4

### Operating System

windows 11

### Printer

kobra

### Describe your model

not water tight

### Add your .zip here ⬇️

https://we.tl/t-j5gyR1Sgsw

thanks for looking, I bought a printer just  for this robot, not going well, I really like your software!

GregValiant, 2023-08-01:
Thanks for the report and the download.
I downloaded the file and the project shows as 436mb.  When I open it in Cura I receive a "No models in the file" message.  Your printer shows up, but no model.
There is a persistent bug that has been reported - have you tried moving and/or rotating the model in Cura?  Sometimes small changes to the Initial Layer Height can get a model to slice.

natennd, 2023-08-01:
Hello, I tried several other slicer/3d programs and nothing will open the
project file saved. I tried every which way nothing seems to help.  I have
the whole thing zipped if you would want to see it, has one stl which was
what I uploaded as project and the rest of it in OBJ