Ultimaker/Cura #16346: Slicing misses all detail and internal geometry - Bad Slicing

PhilBaz, 2023-08-01:
### Cura Version

5.4

### Operating System

win 10

### Printer

Voron 2.4

### Reproduction steps

Slice file.

Slicing misses all detail and internal geometry. Smartavionics branch produces expected/correct result. Ill include images for comparison.

Cura 5.4
[image]
&
[image]

Smartavionics
[image]

### Actual results

[image]

### Expected results

[image]

### Add your .zip and screenshots here ⬇️

[bad slice detail lost.zip](https://github.com/Ultimaker/Cura/files/12232107/bad.slice.detail.lost.zip)

GregValiant, 2023-08-01:
Thanks for the report and the file.
This is in 5.4.0 with the Line Width at 0.40 and "Print Thin Walls" turned on.
[image]

5.X UM Cura is different and the Arachne Engine means some settings have to change.  SmartAvionics fork doesn't have the Arachne Engine.
Is that what you are looking for?
I don't see a bug here.  UM Cura 5.X needs to be set up slightly differently.

PhilBaz, 2023-08-01:
@GregValiant 

Ahh, sorry, I thought I had tried that...

PhilBaz, 2023-08-01:
@GregValiant 

Still the slice with .5 and thin walls off is bad...lots of nonsense in there.

GregValiant, 2023-08-01:
With the Arachne Engine there is no "upper limit" on line width, just a lower limit.  You need to drop the "suggested" line width to at .40.  The Engineering line width of .37 (for a .4 nozzle) might be better suited to this model.  Leonardo didn't paint the Mona Lisa with a house brush.

This is what I'm getting for layer 1
[image]

Layer 2
[image]

Those are wrong?  It's really hard to make out what you have going on within the part.

PhilBaz, 2023-08-01:
@GregValiant 

Thanks Greg, i didn't realize how line width worked under Arachne. Engineering line width, ill have to look into that. 

Its not my part, but yes they are trying some clever tricks. The second layer you just posted isn't quite right, I think. But the one you posted above is, with veins going down the length of the center. 

But, thanks this definitely solves the problem in terms of getting this thing printed as intended. Ill keep this aspect of the workflow in mind going forward.

GregValiant, 2023-08-01:
This is not about your post but it seems like people want all sorts of options to do this that way and that this way.  The program ends up complicated with a lot of settings that can interact in unexpected fashions.
In addition, Cura is highly customizable which adds another layer of complexity.  There is no "Easy" button or built-in "AI" to make the decisions and figure everything out for a user.  It's a complicated app with a long learning curve.  The documentation on how to use some of the settings seems spread out and hard to find. 
I still use 4.13.1 for some stuff and I use SmartAvionics version as it is better on sprialize.  Gotta use the right tool for the job.

Good luck with that piece.  I did slice it with a 0.25 nozzle.  The 0.4 should work fine with the lower line width.

I'll leave this open and the bug label in place.  That model presents a unique printing problem.  Let me know how the print turns out.

PhilBaz, 2023-08-01:
@GregValiant 

Hey Greg, yeah, totally.  The model represents a complex problem and I get there is no easy way to evaluate what the user is trying to achieve in these types of situations.

In the end the program was capable of producing the desired output. So, then user knowledge was the limiting factor. 

Its true, these are highly complex problems and complex sophisticated software. 

I was able to achieve the desired result, in the end essential was 0.4 line width and a first layer height under 0.28. 

Thanks for your help.

GregValiant, 2023-08-01:
Ta-Daa.  Another head scratcher figured out.  I'll go ahead and close this out as a "settings issue" then.

closed 2023-08-01 by GregValiant