Ultimaker/Cura #16347: Slicing failed, ender 3 v2 Neo player

TheSoumilArora, 2023-08-01:
### Cura Version

5.3.1

### Operating System

MacOS Sonoma

### Printer

Ender 3 V2 Neo

### Name abnormal settings

slices successfully using the standard quality default profile

### Describe model location

Centre

### Describe your model

headphone stand with batman logo

### Add your .zip here ⬇️

[project file.zip](https://github.com/Ultimaker/Cura/files/12233883/project.file.zip)

GregValiant, 2023-08-02:
Thanks for the report.
It sliced for me as it is in the project file.
There is a bug that keeps models from slicing.  There are three workarounds:  Move the model, rotate the model, change the Initial Layer Height.
I'll mark this as a duplicate and the Cura team will take a look.

BTW - the supports that grow to the bat wings are really thin.  They're going to fail.  You might want to try "Use Towers" instead of Conical supports.
This would be way easier to print in 3 or 4 pieces and then glued together.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic