Ultimaker/Cura #16350: Ultimaker Cura 5.4 is not displaying materials when selected in settings

GoguyT3D, 2023-08-01:
### Cura Version

5.4

### Operating System

Windows 11

### Printer

Ultimaker S3

### Reproduction steps

Immediately after installing 5.4 from 5.4alpha nightly version I opened cura to see if my personal custom profiles, materials and settings were still in tact. When clicking the dropdown menu from the top of the window and selecting manage materials, the window pops up although the window is not displayed correctly nor is the materials page at all. When clicking on something like profiles the page displays but the title does not show correct text, like two words are on top of each other, also the buttons on the top right of the page dont seem to "mind their own space" and overlap even when maximising the profiles window.
[image]
[New folder.zip](https://github.com/Ultimaker/Cura/files/12234090/New.folder.zip)
Config folder in zip

### Actual results

the correct screen (materials) is not displayed and the profiles screen is displayed with a skewed title and buttons that arent separated.

### Expected results

The material page with my standard and custom materials should show, the title should say "Materials" and the buttons on the top right should be separated with a small margin.
[image]
[image]

### Add your .zip and screenshots here ⬇️

[New folder.zip](https://github.com/Ultimaker/Cura/files/12234115/New.folder.zip)

GoguyT3D, 2023-08-01:
UPDATE!
First, I created a backup with the Ultimaker Cura 5.4, then I shut down my computer, Reopened Cura and the problem persisted. Next I went into Marketplace and downloaded the first random material and installed it(to try to wake up my materials and jog curas memory so to speak), shut down cura (Picture 1) and restarted Cura. Now the Materials screen displays fine and I can switch between printers and Profiles, and materials and see them all with no distorted words at the top. It seems like the graphic of flashing between "Profiles" and "Materials" was staying fixed so two words were printed onto each other, and before the materials screen was not even loading. 
[image]
Cura seems to be performing normally now. (Picture 2)
[image]

closed 2023-08-01 by GoguyT3D

MikeOxawopper, 2023-10-21:
Having the same issue in windows 10 pro with the materials panel and preference panel.

[image]
[image]
[image]

GoguyT3D, 2023-10-21:
@MikeOxawopper try my August 1st comment it solved it for me