Ultimaker/Cura #16351: No Slice facility no blue slice link Cura 5.4

einsvonzehn, 2023-08-01:
### Cura Version

5.4.0

### Operating System

Win 10 64bit

### Printer

Anycubic Kobra

### Reproduction steps

I want to slice but no blue slice link

### Actual results

[image]

### Expected results

I would expect a slice if there was a blue link

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-08-02:
Thanks for the report.
This is the first report I've seen about that behavior.
@MariMakes of the Cura team might have some information about what is going on.

MariMakes, 2023-08-02:
Hey @einsvonzehn,

Thanks for your report 👍
That indeed looks off 🤔 

It seems like parts of your screen have been cut off, like the bottom of your screen where the slice button lives.
What happens if you press the windows key and the arrow up a couple of times? 
It should resize your window, hopefully making the slice button appear. 🤞 

Can you let us know if this works for you?
If not just let us know and we'll take another look.

einsvonzehn, 2023-08-02:
Hallo und Danke für Ihre schnell Nachricht.

Ich habe das Taskleite auto ausblenden und Ja ich siehe das Slice button.

Mfg.

Colin Finch.

 
 

Gesendet: Mittwoch, 02. August 2023 um 10:55 Uhr

MariMakes, 2023-08-03:
Hey @einsvonzehn,

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
I'll be closing this issue. I hope you understand.

closed 2023-08-03 by MariMakes