Ultimaker/Cura #16352: 5.4.0 Slicing Failed with an unexpected error

csterlin, 2023-08-01:
### Cura Version

5.4.0

### Operating System

Windows 11 Home, 22H2, 22621.1778, Intel(R) Iris(R) Xe Graphics

### Printer

Creality Ender 3

### Name abnormal settings

Yes, and yes.

### Describe model location

Yes. The position on the bed had no effect. Rotating in x-y did fix it when adjusted to align with y, but not when aligned with x.

### Describe your model

Yes and yes. Used 3d builder to repair; meshtools could not fix it. Now says it's watertight.

### Add your .zip here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/12234527/cura.log)
[Hudson_Hornet_Duplo_Body_Right.zip](https://github.com/Ultimaker/Cura/files/12234541/Hudson_Hornet_Duplo_Body_Right.zip)
[cura.cfg.zip](https://github.com/Ultimaker/Cura/files/12234553/cura.cfg.zip)

GregValiant, 2023-08-02:
Thanks for the report and the files.
I can confirm all of the above.
I'll mark this as a duplicate slicing error and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic